d4rken-org/capod #1: Not working

chiragnahata, 2022-04-27:
The app is not working. It is not detecting my Redmi earbuds 2c

chiragnahata, 2022-04-27:
362b67c1-a3d6-4a8d-8fdb-1de73adcc9a1 my install id

chiragnahata, 2022-04-29:
> Do the following:
>
>    - Enable Settings > Show all devices
>    - Enable Settings > Debug Settings > Unfiltered BLE Data
>
> Check if your earbuds showup, if they do, decode the Bluetooth data and
> I'll add support.

Done all these steps. But still it is not working

Chirag Nahata

chiragnahata, 2022-04-29:
Now what to do ? Please help

chiragnahata, 2022-04-29:
Log file

chiragnahata, 2022-05-08:
Not resolved yet

d4rken, 2022-05-11:
* Your logfile was not posted.
* If nothing shows after doing [this](https://github.com/d4rken/capod-public/issues/1#issuecomment-1112134423) then it's an issue with your device.
* I don't have Redmi earbuds 2c. I made this app to solve my own itches, it works perfectly with my Pixel 5 & AirPods Pro. I'm open for more, but if it's a totally different device, you'll have to step up and tell me how to decode it :slightly_smiling_face: .

closed 2022-09-22 by chiragnahata