d4rken-org/capod #2: Update UI for Popup

Jomyy, 2022-04-29:
that would be cool and needed....
also would you mind opensourcing the app, i would gladly help with the design

btw this is by far the best app for airpods i tried, except for the ui!

d4rken, 2022-04-29:
> btw this is by far the best app for airpods i tried

:heart: 

> except for the ui!

Yeah, not untrue. I'm more of a `functionality over form` kinda guy :grin:.

> i would gladly help with the design

In what ways?

> also would you mind opensourcing the app

I am contemplating it. Basically this was the idea all along, just don't know when.
I dread the copy cats. I don't mind forks and improvements,
But I have so much hate for people who reupload open source apps to Google Play, just with added ads :rage: . 

I've experienced this with other apps of mine, and seen e.g. @adolfintel from [OpenPods](https://github.com/adolfintel/OpenPods) also struggle with this. It's just immensely demotivating...

Jomyy, 2022-04-29:
Haha ok thats understandable...

I could mock up some designs

Jomyy, 2022-04-29:
also maybe the android 12 color system could be used for more "native/stock" app feel

d4rken, 2022-08-23:
@Jomyy FYI: The app is open-source now. :wink: