d4rken-org/capod #6: Auto pause/play doesn't work with one pod mode on

banna227, 2022-06-03:
Have 2 pods in with 1 pod mode on and auto pause/play doesn't work. I can see the icon that says in ear and the icon goes away when I take the pod out but music does not stop. Works fine with one pod mode off but I don't want to have to switch everytime I use just one AirPod.

d4rken, 2022-06-07:
There is a debug log option in the settings, can you record a log of the issue occuring?

flippz, 2022-08-01:
Have the same problem. The case is that it should stop music when u remove 1 pod when wearing 2 pods Even if in 1pod mode. This is the iPhone way i think

d4rken, 2022-08-01:
The feature works by observing the data changes.
Does the list item show that one pod is removed when your remove it?
If yes then this is a bug that can probably be fixed.
If no, then :shrug:

flippz, 2022-08-01:
Yes it does but it does not pause.
If i remove both it pauses 
(when i have one pod mode activated)

d4rken, 2022-08-23:
This is working as expected for me.

In two-pod mode:
* I have to have both pods in my ear to start music.
* If i remove one pod and still have one pod in my ear it pauses music
* If i have one pod in my ear and insert the second one it continues the music

In single-pod-mode:
* Inserting or removing one pod triggers play/pause.

closed 2022-08-23 by d4rken