d4rken-org/capod #12: Download link in Readme leads to empty page

IzzySoft, 2022-07-24:
… as there are no releases, and no APKs attached to the (not existing) releases. As the loink to Play works, maybe you could add tag & APK here as well?

d4rken, 2022-07-24:
Hey 👋 
Still working on it.
CI should be setup now
Still checking the code for old links.
Also trying to set up Github Sponsors and link to it from the app instead of Google Play IAP.

d4rken, 2022-07-25:
Alright, CI is doing it's thing, let's see if the build scripts work.

https://github.com/d4rken-org/capod/actions/runs/2733521850

Pretty sure that GitHub release will work, hoping the Google Play release also works :sunglasses:.

Does the metadata setup work for you?

IzzySoft, 2022-07-25:
As usual: `full_description.txt` will need an ampty line [before the bullet point list](https://github.com/d4rken-org/capod/blob/main/fastlane/metadata/android/en-US/full_description.txt#L4) (or it would become a "wall of text"). Then only the APK is missing at the release tag :smile:

d4rken, 2022-07-25:
I adjusted the description (though after I tagged the release). 

The last CI worked for GitHub:
https://github.com/d4rken-org/capod/releases

Google Play failed (I jinxed it :grin:), trying again currently.

Do you know why it is not showing the "Releases" section on the project's main page?

d4rken, 2022-07-25:
Ah it's not attaching the APK :thinking:

d4rken, 2022-07-25:
ouch, 7 tries, but finally all is looking good :smiling_face_with_tear: 

https://github.com/d4rken-org/capod/releases/tag/v1.3.13-rc7

[image]

IzzySoft, 2022-07-25:
OK, thanks – added as usual. Issues left:

* `full_description.txt`: the issue mentioned above also applies to all other locales (just manually fixed it for `de-DE` here now)
* `versionName` is truncated, missing the  `-rc7` suffix of the tag name. This means it would pull the very same APK day after day, assuming it's a newer version, only to find out it's not. Possible approaches:
    * include the `rc` suffix with `versionName`
    * utilize `versionCode` with the tag name
* Bugsnag again. Same procedure as usual? Would be easier if it weren't included then to start with :wink:

d4rken, 2022-07-25:
> OK, thanks – added as usual. Issues left:
> 
> * `full_description.txt`: the issue mentioned above also applies to all other locales (just manually fixed it for `de-DE` here now)

Hm will have to update it in sources, pulling from crowdin will overwrite it all again.

> * `versionName` is truncated, missing the  `-rc7` suffix of the tag name. This means it would pull the very same APK day after day, assuming it's a newer version, only to find out it's not. Possible [...]

Not sure I fully understand it, can you elaborate a bit more?

> * Bugsnag again. Same procedure as usual? Would be easier if it weren't included then to start with wink

Currently it's opt-out for gplay and opt-in for foss builds. I will think again about not including it, but receiving combined crash reports from all builds is quite convenient. Getting me a crash stacktrace is not easy for everyone.

d4rken, 2022-08-23:
I think we got this now?

closed 2022-08-23 by d4rken

IzzySoft, 2022-08-23:
Uh, umpf… looks like I missed a notification there… So:

> can you elaborate a bit more?

My updater compares the `versionName` of the app against the tag name (as the latter refers the former). So both should show the same grade of completeness. I just had a little error-of-thought in my above conclusion: as a release-candidate (rc) is considered to precede a release (no-rc), having `1.2.3` in the repo would mean all `1.2.3-rc` tags would be ignored – e.g. an `-rc8`, as it's considered a "lower version" and not an update.

d4rken, 2022-08-24:
I see, I'll take care to better follow semver on the next release.