d4rken-org/capod #14: Google pixle buds support?

gun4qmm7h, 2022-07-27:
Can you add google pixle buds?

d4rken, 2022-07-27:
I can't because I don't have Pixel Buds, if someone that has wants to provide code, then we could add it :slightly_smiling_face:.

Do they transmit information via BLE advertisements like AirPods do?

gun4qmm7h, 2022-07-27:
How can I get there code?

seniorm0ment, 2022-09-22:
I would be very interested in seeing Pixel Buds Pro support.
I would like to be able to replace the Pixel Buds app from Google, with a libre alternative. As long as it has the features to customize gestures, and any other features that the Buds app has thatd be great.
The only downside I see is the inability to get firmware updates. But maybe they publish that publicly it can be grabbed from, I'm unsure.