d4rken-org/capod #15: WF-1000XM4 Support

M-505, 2022-07-28:
`2022-07-28T15:17:03.011Z  V/CAP:Bluetooth:Manager2: Bluetooth event (intent=Intent { act=android.bluetooth.device.action.ACL_CONNECTED flg=0x5000010 (has extras) }, extras=Bundle[{android.bluetooth.device.extra.DEVICE=F8:4E:17:42:F3:11}])
2022-07-28T15:17:03.014Z  D/CAP:Bluetooth:Manager2: Adding F8:4E:17:42:F3:11 to current devices []
2022-07-28T15:17:03.016Z  V/CAP:Monitor:Worker: MonitorMode.onEach(): (ALWAYS, [])
2022-07-28T15:17:03.016Z  D/CAP:Monitor:Worker: Monitor mode: ALWAYS
2022-07-28T15:17:03.018Z  V/CAP:Bluetooth:BleScanner: onScanResult(delay=22ms, callbackType=1, result=ScanResult{device=CC:79:80:67:CF:A0, scanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=[0000fd82-0000-1000-8000-00805f9b34fb, 0000fe03-0000-1000-8000-00805f9b34fb], mServiceSolicitationUuids=[], mManufacturerSpecificData={301=[4, 0, 2, 2, -80, 50, 3, 1, -50, 113, -2, -117, -3, 16, 3, 0, 51, 0, 0, 0]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=LE_WF-1000XM4`

Let me know what else needs to be done, so we can work on this together and get this supported, thank you

d4rken, 2022-07-28:
The main data source for apple devices is the manufacturer specific data in the BLE scan records.

Sony seems to have something too:
mManufacturerSpecificData={301=[4, 0, 2, 2, -80, 50, 3, 1, -50, 113, -2, -117, -3, 16, 3, 0, 51, 0, 0, 0]}

I have no idea what it means though at that's what we have to figure out.

M-505, 2022-07-28:
Well any clue on where I could start would help me, unless you are telling me apple has that in its specs data? But I'm sure to look at it :)

d4rken, 2022-07-28:
No apple didn't explain anything, some people figured it out by (see readme of this repo).

A starting point could be to change something like charging/not charging and see what data changes.

Maybe trying to convert the interger values to hex values.

d4rken, 2022-07-28:
The ByteArray `[4, 0, 2, 2, -80, 50, 3, 1, -50, 113, -2, -117, -3, 16, 3, 0, 51, 0, 0, 0]`
in hex is `04 00 02 02 B0 32 03 01 CE 71 FE 8B FD 10 03 00 33 00 00 00`.

Some random example, for Apple, the Battery value is `FF` when both are full, one `F` for each pod. `EE` would be both pods at like 90%.

closed 2025-05-14 by d4rken