d4rken-org/capod #18: Persistent notification

GrantSavill, 2022-08-03:
Would it be possible to use a different notification category for when there are no devices connected, so this can be switched off if preferred?

d4rken, 2022-08-04:
Hm, this is a bit tricky because the notification is necessary to keep the app alive if features like "force connection" is enabled, otherwise we can't monitor the status in the background.

You are right though that a split notification could work. If you disable the on-going notification in the system, I think the app still keeps the foreground priority :thinking: 

/cc #3

surfaceflinger, 2022-08-21:
Small offtopic, but it's still about notifications:

I have CAPod notifications set as silent so I don't have CAPod icon constantly on status bar, the issue is that when CAPod notification is minimized it shows no device even when AirPods are connected.

Another suggestion would be to use dark icons on the notification when light theme is in use.

https://user-images.githubusercontent.com/44725111/185802537-5ce03daf-9a1a-4e1c-b31a-f25efff3672f.mp4

Also, wouldn't it be possible to have not minimized notification but also not show it on status bar? Maybe force it somehow to the end of the status bar and use blank icon? No idea.

PR #284 (2025-05-14): Add option for separate connected notification
Closes #28
Closes #18

[image]

[image]
commit eb8a9b3b (2025-05-14): Add option for separate connected notification

Closes #28
Closes #18
closed 2025-05-14 by d4rken