d4rken-org/capod #19: Realme Buds Wireless 2S

meanindra, 2022-08-05:
Can you add `realme Buds Wireless 2S`?
Here is some info about this device (linux desktop screenshot)

[image]  

If this is not helpful, can you tell me how you or other extracted Bluetooth data for other devices?
Thanks

d4rken, 2022-08-05:
AirPod data is transmitted through Bluetooth low energy advertisement .

In CAPod enable "unfiltered BLE data" and see if you can pick out your headphones, if they are visible, figure out what the raw data means 🙂

meanindra, 2022-08-05:
My headphones is not visible. Closing this issue.

closed 2022-08-05 by meanindra