d4rken-org/capod #28: Hide "No device" notification

GEkuL, 2022-09-13:
Hi there!
I would like to hide the notification of "No device" since it's pretty useless and spams the space on among notifications.

The trouble is that if I hide it with Android (12) settings, I'll hide the notification when pods are connected as well.

I guess that this notification is essential for the app staying in background to monitor BLE status.

So my suggestion is to divide notifications:
- to stay in background
- to show the info about pods (a.k.a. Device status)

The 2nd one will appear only when the pods are connected. While another one could be hidden via basic android tools. Seems like a win-win strategy.

What do you think?

GEkuL, 2022-09-13:
However the suggestion can lead to  [notifications flood issue](https://github.com/adolfintel/OpenPods/issues/136).

d4rken, 2022-09-13:
> The 2nd one will appear only when the pods are connected. While another one could be hidden via basic android tools. Seems like a win-win strategy.

> What do you think?

Won't this only be obvious to "power-users", and casual users will be annoyed by the 2 notifications, not understanding that they have to disable one?

GEkuL, 2022-09-13:
@d4rken 
Guess you can always add a small instruction or a Notification subfolder in settings.

OR
You can keep the default thing as is now, and add the functionality to split notifications.
This will keep the newbies out of annoying trouble while providing mature users with the flexibility and customizability.

After all, I think those users who care about OSS or know about Fdroid, they pretty sure know how to hide notifications (:

d4rken, 2022-09-13:
I'm in favor of this feature, just not super important to my personal use-cases, so rather low priority for my time allocation :beers:.

I have also thought about whether we could create a channel that is deactivated by default, but I don't think it's possible :thinking: 
Making it optional would be a solution. Do you know any other apps that have some nice patterns for this that users understand well? Could we just write in the keep-alive notification that it can be deactivated? Can we put a button in the notification to disable it? Hmm :thinking: 

We keep the current channel:

https://github.com/d4rken-org/capod/blob/753d31159358914fbbd2228b6274d993f4466900/app/src/main/java/eu/darken/capod/monitor/ui/MonitorNotifications.kt#L99-L100

and introduce a new one... then swap the `setForeground` argument...

https://github.com/d4rken-org/capod/blob/753d31159358914fbbd2228b6274d993f4466900/app/src/main/java/eu/darken/capod/monitor/core/worker/MonitorWorker.kt#L90-L103

GEkuL, 2022-09-17:
@d4rken
Hello again,

I found that [Gadgetbridge](https://codeberg.org/Freeyourgadget/Gadgetbridge) has it's General notification channel (to keep the app alive) as well as some other ones, for example to notify on low battery.

It seems like a similar case to our issue, check it up :)

MottyG56, 2023-01-10:
Any update on this? I really would like to have this,
Thanks, amazing app

PR #284 (2025-05-14): Add option for separate connected notification
Closes #28
Closes #18

[image]

[image]
commit eb8a9b3b (2025-05-14): Add option for separate connected notification

Closes #28
Closes #18
closed 2025-05-14 by d4rken