d4rken-org/capod #31: AirPods Pro 2

GrantSavill, 2022-09-20:
I believe the AirPods Pro 2 model is 1120 in hex. Data appears to be the same as all other Pod devices.

d4rken, 2022-09-20:
Can you share a debug log from CAPod that contains data from AirPods Pro 2? Or just post a few sample hex strings?

GrantSavill, 2022-09-20:
Unfortunately I don't own any, just saw some out in the wild and assumed this must be them:

76: 07 19 01 11 20 2B 88 OF 00 11 05 77 7B OE E5 B8 AD 03 57 2F FA AA 2C F5 65 0F 52

GrantSavill, 2022-09-20:
Just realised they're not even out yet... I have no idea what that could be then.

d4rken, 2022-09-20:
> 76: 07 19 01 11 20 2B 88 OF 00 11 05 77 7B OE E5 B8 AD 03 57 2F FA AA 2C F5 65 0F 52

Did you use image recognition? There some O instead 0 in there. Fixed:

```
07 19 01 11 20 2B 88 0F 00 11 05 77 7B 0E E5 B8 AD 03 57 2F FA AA 2C F5 65 0F 52
```

d4rken, 2022-09-20:
I think this is to vague to add it just yet, but I've created a draft with the necessary classes already in #32. We just need some more data to confirm how the data looks, so we'll have to wait until they are out and someone owns them :smile:.

Your sample decodes with these values:

https://github.com/d4rken-org/capod/blob/3c5dd9f7b7375da84335676e5fb190c2edb966ff/app-common/src/test/java/eu/darken/capod/pods/core/apple/airpods/AirPodsPro2Test.kt#L13-L37

GrantSavill, 2022-09-20:
Yep fair enough. Sounds logical to me :)

d4rken, 2022-09-20:
Some beats devices like the Power Beats Pro also have known colors, or it's a new color that CAPod doesn't know.

Maybe there is a new beats device out that CAPod doesn't know...

GrantSavill, 2022-09-20:
Okay safe to assume they're not AirPods Pro 2 then. Likely Beats Fit Pro, they have new colours.

d4rken, 2022-09-20:
Well, let's also add support for them if we can find someone who can confirm the data, #33 :grin:.

Kuhgelfische, 2022-09-23:
@d4rken Is this the data you need?

2022-09-23T23:16:37.935Z  I/CAP:Pod:Factory: Pod created: UnknownAppleDevice(identifier=Id(id=87990e13-5b26-46ce-a548-aae4d7f2bd84), seenLastAt=2022-09-23T23:16:37.930Z, seenFirstAt=2022-09-23T23:15:18.118Z, seenCounter=69, scanResult=BleScanResult(-37, 5B:D4:E7:12:A1:F6, 12907308088809, 76: 07 19 01 14 20 55 88 F9 51 00 04 20 50 03 CA D5 C9 AC 0F FA 84 78 94 5A 4D DF F5, proximityMessage=ProximityPairing.Message(type=7, length=25, data=01 14 20 55 88 F9 51 00 04 20 50 03 CA D5 C9 AC 0F FA 84 78 94 5A 4D DF F5), confidence=1.0, rssiAverage=-40)

d4rken, 2022-09-24:
@Kuhgelfische Yes, awesome, specifically this:

```
07 19 01 14 20 55 88 F9 51 00 04 20 50 03 CA D5 C9 AC 0F FA 84 78 94 5A 4D DF F5
```

that would make the AirPods Pro 2 identifier `14 20`.  It would decode as:

* Left pod is the microphone
* Both pods at 80% battery
* Both pods are charging
* Case at 90% battery
* Case is charging
        
Does that fit?

https://github.com/d4rken-org/capod/blob/5cd750a9210d8b0098a209b1256868f5af0409dc/app-common/src/test/java/eu/darken/capod/pods/core/apple/airpods/AirPodsPro2Test.kt#L28-L42

d4rken, 2022-09-24:
I think we've got it, will make a new update. :partying_face:

closed 2022-09-24 by d4rken