d4rken-org/capod #34: i10, i11, i12 supoort

Relected, 2022-09-21:
these airpods clones are suprisengly popular because they are cheap off brands that offer decent quality audio, touch control, and they use BT 5.

if they could be supported then I would like them to be added

d4rken, 2022-09-21:
If they are close to AirPods, then they should be easy to add, just need someone to supply some data :slightly_smiling_face:.

Relected, 2022-09-21:
@d4rken they are VERY close to airpods in terms of functionality and design

also how can I provide data?

d4rken, 2022-09-21:
Record a debug log, see settings. If they don't show as unknown device, enable unfiltered BLE data first.

I need to know the device state, i.e. they are fully charged and charging + the message they broadcast to match the values.

closed 2025-05-14 by d4rken