d4rken-org/capod #35: notification can't allowed

OrangeOC, 2022-09-22:
When I update this app to version V2.1.1-rc0, the notifications themselves are no longer allowed.
I'm using pixel5a on android13.In the FOSS version you can allow notification permissions as usual.

d4rken, 2022-09-22:
This should have been fixed in `v2.1.1-rc0` via #29.
Are you sure you have the latest version on Google Play? You have to opt-in to beta testing, it's not released to production on Google Play.

[image]

OrangeOC, 2022-09-22:
Oh,sorry!
I misunderstood the version.
After running the update, it worked as usual!

closed 2022-09-22 by OrangeOC