d4rken-org/capod #39: Is it possible to add support for the loudspeakers in the Airpods pro 2 case?

Corlos2078, 2022-10-01:


d4rken, 2022-10-01:
No one knows how to do it :face_exhaling:.
If someone figures it out, I'd be happy to add it.

K3NOXOFFICIAL, 2024-01-16:
Hasn't read and write operations been figured out by someone recently? Don't really know anything specific but might be a good idea to look into.

xerootg, 2024-07-11:
This code popped up recently for setting airpods modes for windows, maybe that's a good starting point? https://github.com/bho3538/Airpods-Windows-Control

d4rken, 2024-07-12:
Yes, the magicpods project has figured out read/write support. I could reproduce it on my desktop computer but not on Android. As far as I know, Android does not support the special Bluetooth connection type that this requires, at least i could not figure it out.

d4rken, 2024-07-12:
Requires #215

kavishdevar, 2025-05-14:
Actually, we don't need #215. As I've mentioned before in https://github.com/steam3d/MagicPodsCore/issues/14, the AirPods case host a GATT server. This is how the sound is played, and battery status is retrieved even when they're not connected. But, to tell it do something, we probably need to authenticate with it. Unfortunately, I haven't figured it out exactly out this "authentication" happens, and whatever I write to the characteristic is ignored by the case.