d4rken-org/capod #59: Airpods Pro Gen 2 Support Help

easyax, 2022-12-04:
I just bought airpods pro 2nd generation and I am an android user. I did not fully understand the application.  It says in ear even though I take off the earphones. And when the headphones are connected, the popup does not appear.
[Screen Record ](https://drive.google.com/file/d/1eOqr38dC1_wm4s0AfiV4NjuipxghZciI/view?usp=drivesdk) 
[image]
[image]

d4rken, 2022-12-04:
The in-ear detection is not working right for Airpods Pro 2, likely a bug in the firmware (see #38).

For the pop-up to work, do you have your device type and address selected in settings?

easyax, 2022-12-05:
After resetting and reconnecting airpods pop up solved. Is there nothing I can do to use spatial sound?

d4rken, 2022-12-05:
> Is there nothing I can do to use spatial sound?

Apple has not made this accessible :frowning_face:

closed 2022-12-05 by d4rken