d4rken-org/capod #62: [Feature Request] AirPods icons in Notification

christoschronopoulos, 2022-12-21:
Thank you for your efforts! I know it is trivial and solely aesthetic, but I would like to ask if it is possible to edit the notification icons to the actual images of the AiPods (e.g., Pro 2) and Case.

d4rken, 2022-12-22:
Yes it's possible, can you can design a good looking set?

christoschronopoulos, 2022-12-22:
I can definitely do that! Now that it's Xmas holidays I will have some free time! 
Only, please let me know what are the filetype and image size requirements.

d4rken, 2022-12-22:
Can you show me a screenshot of which icons you would like to replace, to make sure we are talking about the right ones :grin:

christoschronopoulos, 2022-12-22:
Sure, please see the screenshots below (icons marked in red)! I am primarily talking about the notification icons, but if we could also change the ones in the widget would be great!
[image]
[image]

christoschronopoulos, 2022-12-22:
I would very much like to replace those icons with actual images of the AirPods Pro 2. Something like that 
[image]
[image]

d4rken, 2022-12-22:
Most of the current icons are from https://materialdesignicons.com/
That's also the style i use all over the app, but I'm not against using less "icon-like" images if it looks good :smiley:.

A vector graphic would be best, i.e. a SVG file. This is the SVG of the current icon:
https://github.com/d4rken-org/capod/blob/main/app-common/src/main/res/drawable/ic_airpod_left_24.xml

A PNG is also possible, needs to be least 48px*48px.
Transparent background.

> I would very much like to replace those icons with actual images of the AirPods Pro 2. Something like that

Just make sure no Apple lawyers descend on us like banshees.

> but if we could also change the ones in the widget would be great!

We can, if we get a full set we can use to display the specific icon everywhere, we'd need these 4 types of image:
[image][image]
[image][image]

christoschronopoulos, 2022-12-22:
> Just make sure no Apple lawyers descend on us like banshees.

That's a valid point! Not sure though if using them as 45px icons in an open source app would be an infringement. :/

> A vector graphic would be best, i.e. a SVG file. This is the SVG of the current icon: https://github.com/d4rken-org/capod/blob/main/app-common/src/main/res/drawable/ic_airpod_left_24.xml
>
> A PNG is also possible, needs to be least 48px*48px. Transparent background.

In the attached zip I have included both SVG and PNG images at 500px. Regarding the tint, I don't know how the white icons would look in Dark/Light themes. Could you try and share a screenshot or two?

[AirPods Pro 2.zip](https://github.com/d4rken-org/capod/files/10290043/AirPods.Pro.2.zip)

d4rken, 2022-12-23:
With tintmode multiply
[image][image]

Tintmode `src_atop`

[image][image]

d4rken, 2022-12-23:
Building a new release right now, see how you like it.
I went with tintmode multiply.

btw, the SVGs are not "real" they just contain the png image in binary form.

christoschronopoulos, 2022-12-23:
Thanks for your effort!!! Hmmm, tbh I don't like how they look. Maybe it is due to the low quality of the icons that I sent when their size is reduced. Good to know about the SVGs. I exported them in Photoshop, so I didn't expect that. I need to find another way to create them, maybe GIMP will do the job. 

I don't know what "tintmode multiply" is. Could you please explain?

d4rken, 2022-12-23:
> I don't know what "tintmode multiply" is. Could you please explain?

Just different ways to apply the tint color, you can see the difference in the examples above.
[image][image]

christoschronopoulos, 2022-12-23:
[AirPods Pro 2_svg.zip](https://github.com/d4rken-org/capod/files/10297189/AirPods.Pro.2_svg.zip)

I worked a little bit on the SVG icons with Inkscape. I think they are "true" SVGs now, and better in terms of quality. How can I get a new release to try it out? Will it affect the Pro version that I have?

d4rken, 2022-12-24:
> How can I get a new release to try it out?

The previous icons are in the latest release, to get these newer ones, you could copy the current icons in and build a new version from sources.

> Will it affect the Pro version that I have?

You will have to uninstall that but can later reinstall.

I don't see a difference to the previous icons :thinking: The left/right icon could need some extra padding/margins.

[image][image]

d4rken, 2022-12-24:
The real vector graphics are too complex, Android Studio warns about performance issuse:
>  Very long vector path (12290 characters), which is bad for performance. Considering reducing precision, removing minor details or rasterizing vector.

There are also some artifacts:
[image]

d4rken, 2022-12-24:
I've added some padding. Looks good imho.
[image]
[image][image]

Noticed a bug in the current release, the new icons are only used for AirPod Pro1 not the Pro2.

christoschronopoulos, 2022-12-24:
> I've added some padding. Looks good imho. [image] [image][image]
> 
> Noticed a bug in the current release, the new icons are only used for AirPod Pro1 not the Pro2.

Looks good indeed! I will check the latest release with my AirPods Pro 2.

christoschronopoulos, 2022-12-24:
@d4rken is there any way to prevent the tint color from being applied to the icons? I mean, I would like to test the icons appearing as white (original color) always.

d4rken, 2022-12-24:
> is there any way to prevent the tint color from being applied to the icons? I mean, I would like to test the icons appearing as white (original color) always.

Not in the build APK, you'd have to edit the code and install from source.

[image][image]

Without tinting the contrast is low on bright backgrounds

christoschronopoulos, 2022-12-24:
I see your point, but tbh I prefer the white even on light-color backgrounds as I see it on your screenshot. It looks really good!

d4rken, 2022-12-24:
For in-app this could be viable as there we fixed the theme colors, but the widget and notifications make use of Android's dynamic theming. So the colors match what ever the user has picked (fixed, custom or based on wallpaper).

Tough :thinking: 

I'm tending towards keeping the tinting, the custom icons are already an improvement over the previous ones.

christoschronopoulos, 2022-12-24:
Hmmm, since the new icons look good in your screenshot with (almost) white background, I don't see why they wouldn't look as good with other user-selected light-color themes. But anyway.

d4rken, 2022-12-24:
Maybe if we have icons like these for all devices?
I don't want to add extra code to turn on/off depending on which icon it is.

Let's not over-complicate this. :beers:

closed 2022-12-24 by d4rken

christoschronopoulos, 2022-12-24:
Makes total sense. In the following days I will try to create proper SVGs for the major AirPods versions as per your feedback, and you can take it from there of you think is feasible.