d4rken-org/capod #65: AirPods Pro 2 wrong battery level

ursusursus, 2022-12-26:
App says 90%, yet my Mac says 95% and other android apps say 95% as well. What do you need to repro?

ursusursus, 2022-12-26:
Okay now I notice battery level only 0, 10, 20.. 100, and some apps add 5 to that arbitrarily, to make it look nicer I guess

Can we to better? My mac shows actual level say 93%

d4rken, 2022-12-27:
Difficult...

Maybe Apple uses data we don't have access to, which contains more accurate values.

Or Apple get's same data but tracks it over time to guesstimate values in between.
This is tough for us to do because we can't uniquely identify the AirPods (consistently), while Apple can identify them.
The encrypted data contains this information, but Apple doesn't share how to decode it and no one else has figured it out. 

So any heuristic we do on the time is definitely worse then what you get from Apple.
You can enable the debug mode and then each entry will show how long CAPod managed to keep track of your AirPods.

It works okay if yours are the only one around, but get problematic in crowded places.

someone5678, 2024-06-09:
> Difficult...
> 
> Maybe Apple uses data we don't have access to, which contains more accurate values.
> 
> Or Apple get's same data but tracks it over time to guesstimate values in between. This is tough for us to do because we can't uniquely identify the AirPods (consistently), while Apple can identify them. The encrypted data contains this information, but Apple doesn't share how to decode it and no one else has figured it out.
> 
> So any heuristic we do on the time is definitely worse then what you get from Apple. You can enable the debug mode and then each entry will show how long CAPod managed to keep track of your AirPods.
> 
> It works okay if yours are the only one around, but get problematic in crowded places.
> 
> Adding 5% to each would be no problem, might make sense, maybe 0% isn't viable as at 0% they turn off, so the 0% value should be 5%? 🤔

I know this is an old issue, but recently, MagicPods' developer, steam3d make showing 1% steps work on their Steam Deck Decky Loader module!

[MagicPodsDecky](https://github.com/steam3d/MagicPodsDecky)
[MagicPodsCore](https://github.com/steam3d/MagicPodsCore)

d4rken, 2024-06-17:
Thanks for the ping. I don't think we can copy this easily.

AFAIK this is done by using direct read/write access to the AirPods.

The MagicPods project has decoded some of the Bluetooth communication between iPhone and AirPods and replicated it.

I have looked into it and tried to do this on my computer, and it worked, but the specific mode of communication that we need to do this, is not available in Android.

It needs a specific type/mode of Bluetooth communication channel and specific settings, and this combination is not available on Android :frowning_face:, it would require a custom ROM just for this :skull:.

Happy to be proven wrong here, but I've run into a dead-end with this.

closed 2025-05-14 by d4rken