d4rken-org/capod #67: Add support for Anker Soundcore Liberty 2

Kreuger, 2022-12-27:
Hey I was wondering if you could add support for these buds. I came across [this repo](https://github.com/gmallios/SoundcoreManager) and Im wondering if the info they have in the bluetooth files would be enough for you to add support.

d4rken, 2022-12-27:
Do they work using BLE broadcasts too?

d4rken, 2022-12-27:
My rust skulls are rusty :sweat_smile:.
Maybe @gmallios could chime in: Is the battery data available via BLE broadcasts (scanrecords/advertisements) or does it require a paired device you have to query specific BLE characteristics?

Some exploration:
He gets some ByteArray with data and then decodes it, so far that is what CAPod does too:
https://github.com/gmallios/SoundcoreManager/blob/ee5917608a7ca5adbc7e8fd1af337cbf7eedfcd6/soundcore-lib/src/devices/A3951Device.rs#L248-L272

That data is requested, not sure if this is a BLE query against some characteristic or just a cache?
https://github.com/gmallios/SoundcoreManager/blob/ee5917608a7ca5adbc7e8fd1af337cbf7eedfcd6/soundcore-lib/src/devices/A3951Device.rs#L71-L80

We have send and receive via `RFComm`
https://github.com/gmallios/SoundcoreManager/blob/ee5917608a7ca5adbc7e8fd1af337cbf7eedfcd6/soundcore-lib/src/devices/A3951Device.rs#L55-L69

This looks like connection establishment and service discovery.
https://github.com/gmallios/SoundcoreManager/blob/37987d8ba063f3067e1cb64aaba4ac10ed005408/bluetooth-lib/iobluetooth/src/server.rs#L92-L125

This points to the data being gathered using a 1to1 connection with a paired BLE device and not just listening to broadcasts.

I'm currently not up for implementing that into CAPod. Especially as I don't have any headphones to develop this against. Too much work, too little gain? :thinking:

gmallios, 2022-12-27:
Hey, I'm open to helping and contributing to this project since I have decompiled a lot of the official Soundcore app :). What I am using currently is the equivalent to Android's BluetoothSocket(Think of it like TCP sockets) . It requires the device to be paired and be connected. Some newer devices from Soundcore/Anker do support BLE but as far as I remember Liberty 2s do not.

gmallios, 2022-12-27:
> What information does the Android OS show for these without installing extra apps? Battery level? Or are they like AirPods and no information is available without extra apps?

My Liberty Air 2 Pros report the case battery I think or the average of the 2 earpiece.

d4rken, 2022-12-27:
@gmallios Thanks for chiming in!

I'm not against this feature, I just don't have motivation for it at the moment. My head is already grinding gears about how it could be integrated into the apps architecture :sweat_smile:.
I currently have a few other projects I working on though.

I'm keeping this ticket open, see how much demand there is for this device being supported.

Would probably have to buy one of the "older" SoundCore devices and some of the newer to develop against? :thinking:

Might have to change the CAPod's branding a bit, how would people know to use CAPod for devices not made by Apple :grin:

@Kreuger How did you come to check CAPod for using it with Anker devices?

For Apple devices my motivation was being unsatisfied with existing Android apps.
If the Anker app is good, is there need for another app?
People don't use that many earbuds that all-in-one-app is needed, is it?

gmallios, 2022-12-27:
Yeah, it requires a very different architecture since you have to account for non-BLE devices that require to be connected. I think focusing on BLE hardware is the right path for now. Since I have access to Airpods Pros, I can try to reverse engineer the writing protocol and contribute to this project. 
@d4rken I think you should create like a to-do list to keep track of progress and tasks 😀

Kreuger, 2022-12-27:
> @Kreuger How did you come to check CAPod for using it with Anker devices?
> 
> For Apple devices my motivation was being unsatisfied with existing Android apps. If the Anker app is good, is there need for another app? People don't use that many earbuds that all-in-one-app is needed, is it?

The motivation for me is that I prefer to use open source/libre apps as much as possible. So if I can get rid of their proprietary app, I will.

d4rken, 2022-12-27:
> Yeah, it requires a very different architecture since you have to account for non-BLE devices that require to be connected. I think focusing on BLE hardware is the right path for now. Since I have [...]

:heart_eyes: If you can figure that out, it would be HUGE. I don't know of any app (on any platform, except MacOS) that can do this. Figuring this out would increase priority of implementing #68 a lot.

> @d4rken I think you should create like a to-do list to keep track of progress and tasks grinning

I've created #68 to track implementation of of active data retrieval. If you are working on something e.g. reverse engineering how to write/change settings, feel free to make a ticket to too.

> Edit: Android shows battery level if I access the device through the bluetooth settings menu. I would love to have it shown in the notification bar as well.

Of the case, the buds? each? combined?

Kreuger, 2022-12-27:
>>Of the case, the buds? each? combined?

I think it is for the 2 buds together. The Soundcore app shows individual battery levels.

gmallios, 2023-06-05:
Turns out after all, Soundcore Liberty Air 2 **Pros** support BLE and have 2 Characteristics (Read-Write) and I am able to change ANC modes.

Kreuger, 2023-08-03:
@gmallios Have you made any progress with this?

closed 2025-05-14 by d4rken

Kreuger, 2025-05-14:
It wouldve been great if you had added a comment about why youre closing this. Thankfully, GadgetBridge has the support I needed.

d4rken, 2025-05-15:
Sorry. I don't own these headphones and don't plan to buy them. There was no progress for 2+ years. I decided this is out of scope for the app.