d4rken-org/capod #68: Support for active data retrieval via direct connections with paired devices

d4rken, 2022-12-27:
Currently CAPod obtains data passively.
Supported devices (e.g. AirPods) regularly broadcast BLE advertisements.
CAPod receives these `ScanRecords`'s and data with contains the information we display.
This does not require any pairing.

Most non-apple headphones only offer data via BLE characteristics which can be read if the headphones are paired and connected.

To support devices that only expose information via BLE characteristics, CAPod needs a new mechanism that obtains data from connected devices.

Some brainstorming:
* Already have active worker if device is connected.
    * Flag devices that require active connection
    * Automatically launch worker for these devices (for Apple devices it can stay optional)
* For connected devices establish a connection within CAPod, query services and characteristics
   * Map scan speed to intervals at which we query the device
       * Devices could have BLE notifications enabled, so in addition to manual refresh intervals, code should be reactive.
   * Some sort of caching? 
   * Device modeling (interfaces) should work for this too.
   * Some kind of factory pattern that is created depending on the device type
* Combine data from current BLE data provider with the new one
    * `PodMonitor`  combines the current flow with broadcast data, and the new one
    * Extra connection based monitor lives app wide, worker just keeps it alive?
* Adjust some texts/branding due to upcoming support for non apple devices
* Make this a pro feature?
* With a connection we can write data too
     * Extra settings for devices that support this? Maybe an action button shows in each card that supports this

TBC

d4rken, 2022-12-27:
This is a requirement for:

* Surely #67
* Maybe #34?
* Possibly #15
* Maybe #14?
* Maybe #7

Kreuger, 2022-12-27:
On the subject of a pro version, will you add an unlocker like you have for SDMaid?

d4rken, 2022-12-27:
> On the subject of a pro version, will you add an unlocker like you have for SDMaid?

No. Only did that for SD Maid because IAPs didn't exist, wouldn't have done it if IAPs existed, it's a "legacy" situation now. If I could transfer everyone from Unlocker to IAP without chaos I'd do that.

I'm trying to cut down on the overhead that apps generate besides actually improving the app. Still need money to live though :sweat_smile:.

Trying a straight forward "honor" based system current... CAPod has:
* An IAP for a "pro upgrade" on Google Play
* A "FOSS" build on GitHub where you are asked to donate/sponsor development on GitHub, but you just get the pro features. No idea how well that works currently. Very few people sponsored me on GitHub, but some may have just bought pro on Google Play too and switched to FOSS later?

Comparing IAP purchases on Google Play with downloads on Google Play with Downloads on Github and sponsorships? If paying is optional, people don't pay :frowning:, but I'm willing to experiment with this a bit more.
I think it currently works out because there are only few power users who want the FOSS version and most are not power users and prefer the convenience of Google Play.

Kreuger, 2022-12-27:
Yeah I guess I definitely fall into an advanced user category. I try to avoid using the Play Store if I can, now. I bought SDMaid years ago so if I have to use it again for a purchase, I will. I have an unused balance anyways.

gmallios, 2022-12-28:
Active data retrieval will be needed for Airpods, since after some bluetooth data capturing it uses AAP (Apple Accessory Protocol) which is an L2CAP based protocol.

d4rken, 2025-05-14:
Maybe possible in a future Android version.
Requires a change in the Android firmware, see #215