d4rken-org/capod #78: (request) Google Assistant Support

thefiredragon, 2023-01-22:
A nice to have would be support for the google assistant, also to remap the button action on Airpods to enable Google Assistant
Best regards

Mhsbrian, 2023-04-21:
Great enhancement request, would offer a complete package IMO.

Vcele, 2023-07-30:
I've looked into some other implementations, and it seems like they're detecting it through music playback. If we could receive those play, paused commands it could be possible to trigger the assistant for example by pressing the stem two times slowly on the pros.

Vcele, 2023-07-30:
I am not that experienced in Kotlin but if I find time I'll try to implement this 👍

closed 2025-05-14 by d4rken

saostad, 2025-10-07:
Hi,
Having access to Google Assistant is a great feature, most of the other apps in the Play Store have this feature.
Is there any specific reason this issue is not planned?