d4rken-org/capod #97: Genuine AirPods Gen 3 are detected as fake with seemingly incorrect information

JTriantafylos, 2023-02-02:
My (genuine as far as I can tell) Gen 3 AirPods don't seem to be detected unless I explicitly set the device model in the settings to "AirPods (Gen 3)?" (i.e., fake). Once they are detected they seem to always show the pods as 100% battery and charging, and the case as 70% battery, no matter what.

I have tried a variety of combinations of compatibility settings in the app, but still the only way they were detected was by manually setting the device address and device model.

The debug information shown in the app when they are detected is:

`[76: 07 13 01 13 20 75 AA 37 37 00 10 00 E4 E4 40 00 00 00 00 00 00]`

I also attached a debug log recording where I just opened the case with the AirPods in it.

[capod_logfile_1675365727977.log](https://github.com/d4rken-org/capod/files/10572671/capod_logfile_1675365727977.log)

Let me know if there is any other information I can gather to help solve this, and thank you for all the hard work that has been put into this app!

d4rken, 2023-02-03:
This all looks correct unfortunately.

As far as I know, those are fake AirPods.
In a sample size of 4-5 AirPods where I talked to users on Discord and looked into it; all the fake airpods were detected correctly.

The tell-tale sign is the data being too short. This message is 19 bytes long while the message from genuine pods are bytes long.

> Once they are detected they seem to always show the pods as 100% battery and charging, and the case as 70% battery, no matter what.

The pod batteries are encoded in the `AA` so showing both at 100% is also correct (as in what the data is telling us).

JTriantafylos, 2023-02-03:
> This all looks correct unfortunately.
>
> As far as I know, those are fake AirPods. In a sample size of 4-5 AirPods where I talked to users on Discord and looked into it; all the fake airpods were [...]

Yikes, guess I got played...

Anyways, thanks for the quick response!
I'll close this issue up now.

closed 2023-02-03 by JTriantafylos