d4rken-org/capod #99: AirPod Pro 2 discovery problems and others

p3dr01978, 2023-02-11:
HI, my airpod pro 2 are not recognised if I set my device type correctly in the settings, however they are recognised as mine if I keep unknown in the setting.

None of the pro actions are recognised at all (auto pause etc...)

My airpod firmware is 5b58 model number a2698

d4rken, 2023-02-11:
What code do you get? The long hex number at the bottom of the card. You might have to enable debug mode in CAPod.

Could be fake AirPods 2?

p3dr01978, 2023-02-11:
The discovery issue  I mentioned at the beginning was due to fake air pod 2 (stupid me, but I fixed it few minute after I wrote here ), for the actions like pause removing one arpod, etc... no solution and they are not working. My ID  (see the attached image) 
[image]

d4rken, 2023-02-12:
> The discovery issue I mentioned at the beginning was due to fake air pod 2 (stupid me, but I fixed it few minute after I wrote here ), for the actions like pause removing one arpod, etc... no solution and they are not working. 

This may be due to an Apple firmware bug, see: #38

p3dr01978, 2023-02-12:
thanks let's wait a new firmware

closed 2023-06-27 by d4rken

Adithyan007hack, 2024-02-23:
> The discovery issue I mentioned at the beginning was due to fake air pod 2 (stupid me, but I fixed it few minute after I wrote here ), for the actions like pause removing one arpod, etc... no solution and they are not working. My ID (see the attached image) [image]

Did succeeded in connecting fake airpods??