d4rken-org/capod #122: Add support for Earfun Pro SV

hartigan43, 2023-03-21:
Add support for [Earfun Air Pro SV](https://www.myearfun.com/headphones/earfun-air-pro-sv-anc-true-wireless-earbuds-black)

I'm able to pick them up when setting unfiltered BLE data in debug.  Happy to help implement, just unsure where to start.

hartigan43, 2023-03-27:
Should I be tracking the HEX data somehow within the app debug settings?  Is there a way to get it to strictly listen to a specific MAC?

I enabled debug logging and only received one line for it:

```2023-03-27T17:13:09.177Z  D/CAP:BluetoothEventReceiver: Event related to B0:F1:A3:84:38:EF```

CAPod does detect when I interact with the headphones as it pops to the top of the list of unknown devices.  Filtering the logs down to that MAC It looks like the only change is the value in scanResult:

```2023-03-27T17:23:32.289Z  I/CAP:Pod:Factory: Decoding BleScanResult(-64, FF:FF:FF:84:38:EF, 8847026410125, 564: 21 00 B0 F1 A3 84 38 EF
2023-03-27T17:23:32.289Z  W/CAP:Pod:Apple:Factory: Data contained no continuity messages: BleScanResult(-64, FF:FF:FF:84:38:EF, 8847026410125, 564: 21 00 B0 F1 A3 84 38 EF
2023-03-27T17:23:32.290Z  D/CAP:Pod:Unknown:Factory: searchHistory1: Didn't recognize: UnknownDevice(identifier=Id(id=81820dfd-5a5c-47b1-8a8f-5b0faa0863aa), seenLastAt=2023-03-27T17:23:32.289Z, seenFirstAt=2023-03-27T17:23:32.289Z, seenCounter=1, scanResult=BleScanResult(-64, FF:FF:FF:84:38:EF, 8847026410125, 564: 21 00 B0 F1 A3 84 38 EF, reliability=0.0, rssiAverage=null)
2023-03-27T17:23:32.290Z  D/CAP:Pod:Unknown:Factory: searchHistory1: Creating new history for UnknownDevice(identifier=Id(id=81820dfd-5a5c-47b1-8a8f-5b0faa0863aa), seenLastAt=2023-03-27T17:23:32.289Z, seenFirstAt=2023-03-27T17:23:32.289Z, seenCounter=1, scanResult=BleScanResult(-64, FF:FF:FF:84:38:EF, 8847026410125, 564: 21 00 B0 F1 A3 84 38 EF, reliability=0.0, rssiAverage=null)
2023-03-27T17:23:32.290Z  I/CAP:Pod:Factory: Pod created: UnknownDevice(identifier=Id(id=81820dfd-5a5c-47b1-8a8f-5b0faa0863aa), seenLastAt=2023-03-27T17:23:32.289Z, seenFirstAt=2023-03-27T17:23:32.289Z, seenCounter=1, scanResult=BleScanResult(-64, FF:FF:FF:84:38:EF, 8847026410125, 564: 21 00 B0 F1 A3 84 38 EF, reliability=0.0, rssiAverage=null)
2023-03-27T17:23:34.346Z  V/CAP:Pod:Factory: Trying to create Pod for BleScanResult(-68, FF:FF:FF:84:38:EF, 8849082378978, 564: 21 00 B0 F1 A3 84 38 EF
```

```2023-03-27T17:24:52.483Z  W/CAP:Pod:Apple:Factory: Data contained no continuity messages: BleScanResult(-44, FF:FF:FF:84:38:EF, 8927221565094, 564: 21 00 B0 F1 A3 84 38 EF
2023-03-27T17:24:52.483Z  V/CAP:Pod:Unknown:Factory: searchHistory1: Recovered previous ID via address: KnownDevice(history=20, last=UnknownDevice(identifier=Id(id=81820dfd-5a5c-47b1-8a8f-5b0faa0863aa), seenLastAt=2023-03-27T17:24:51.996Z, seenFirstAt=2023-03-27T17:24:51.996Z, seenCounter=1, scanResult=BleScanResult(-43, FF:FF:FF:84:38:EF, 8926735222959, 564: 21 00 B0 F1 A3 84 38 EF, reliability=0.0, rssiAverage=null))
2023-03-27T17:24:52.483Z  I/CAP:Pod:Factory: Pod created: UnknownDevice(identifier=Id(id=81820dfd-5a5c-47b1-8a8f-5b0faa0863aa), seenLastAt=2023-03-27T17:24:52.482Z, seenFirstAt=2023-03-27T17:23:32.289Z, seenCounter=42, scanResult=BleScanResult(-44, FF:FF:FF:84:38:EF, 8927221565094, 564: 21 00 B0 F1 A3 84 38 EF, reliability=1.0, rssiAverage=-43)
2023-03-27T17:24:53.008Z  V/CAP:Pod:Factory: Trying to create Pod for BleScanResult(-44, FF:FF:FF:84:38:EF, 8927747700407, 564: 21 00 B0 F1 A3 84 38 EF
2023-03-27T17:24:53.008Z  I/CAP:Pod:Factory: Decoding BleScanResult(-44, FF:FF:FF:84:38:EF, 8927747700407, 564: 21 00 B0 F1 A3 84 38 EF
```

Are the `8847026410125` and `8927747700407` portions of the scan the current state of the headphones?

d4rken, 2023-03-28:
For AirPods it's all in the BLE manufactor specific advertising data, in your log that would be `564: 21 00 B0 F1 A3 84 38 EF`.

> CAPod does detect when I interact with the headphones as it pops to the top of the list of unknown devices.

The list sorting is just based on most recent data and signal quality, not data content.

Filtering the logs down to that MAC It looks like the only change is the value in scanResult.
Does the manufactorspecific data field change? In your examples it look like it stays the same.
It's possible they don't use the BLE advertising data, but instead expose an extra service within the Bluetooth audio connection. A specific Bluetooth service/Bluetooth characteristic.

This is a nice app to investigate things:
https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp

closed 2025-05-14 by d4rken