d4rken-org/capod #124: How do I fix this

ranvirsi1, 2023-03-21:
Mine is saying no primary device when I set it up how do I fix it?

d4rken, 2023-03-22:
This is a ROM issue. Try combining different compatibility options. If nothing helps then you are out of luck.
You could try different AirPods with the same phone, or the same AirPods with a different phone.

ranvirsi1, 2023-03-24:
Like what compatibility settimg

ranvirsi1, 2023-03-24:
My device is Samsung Galaxy tab s5e

chubecode, 2023-03-28:
Sam issues with Vivo devices.  Why it is ROM issue, my airpod still get connected with the device.

d4rken, 2023-03-28:
> Sam issues with Vivo devices. Why it is ROM issue, my airpod still get connected with the device.

The audio connection is a different type of Bluetooth technology than the one where the status information comes from. So it's possible that one works while the other doesn't.

d4rken, 2023-03-28:
> Thanks for respond. Can we detect which device will work or not, any tip for that ?

No idea, seems pretty random to me.

d4rken, 2023-03-28:
> Thanks for respond. Can we detect which device will work or not, any tip for that ?

Saw that you are an Android developer :+1: 
If you want to debug this, play around with the options in this class:

https://github.com/d4rken-org/capod/blob/main/app-common/src/main/java/eu/darken/capod/common/bluetooth/BleScanner.kt

On affected devices the callback either does not return or returns empty lists:

https://github.com/d4rken-org/capod/blob/cea8f2290dc11c7426206fd12a74e96998048d2d/app-common/src/main/java/eu/darken/capod/common/bluetooth/BleScanner.kt#L71-L96

On some devices, changing the API call to NOT use batch mode fixes the issue. On others it only works in batchmode.

Reading some  forums and bug reports on Google, there are some devices where the Bluetooth chip doesn't support a specific BLE scan hardware modes and the ROM devs tried to fake it in software and their solution is buggy.

Maybe you can find another workaround?

chubecode, 2023-03-29:
Yap, I already take some debug, I take look into raw data, my device still return other BLE devices( like speaker, miband,etc..)  but without my airpod. I see some app like AssistantTrigger, can popup and see battery info, it works with my device. So it makes me confuse. Maybe they using another technique, without BLE scanner ?