d4rken-org/capod #137: CAPod Pro never find my airpods at first

ceifa, 2023-06-03:
I always have to troubleshoot it to be found :/

Version: Latest on play store
Phone: Galaxy S9
Android: 10
Device: Airpods Pro

[image]

d4rken, 2023-06-04:
Can you define "always"? When restarting CAPod? The phone? Your Pods?

The settings the troubleshooter ends up with are the default settings, so it's weird that this is required. Do the settings save? If you go into app settings, are the saved settings the same as the one in `#8` from the log?

ceifa, 2023-06-07:
> Can you define "always"? When restarting CAPod? The phone? Your Pods?

When I open the airpods case. It shows a notification with "No device".

> The settings the troubleshooter ends up with are the default settings, so it's weird that this is required. Do the settings save? If you go into app settings, are the saved settings the same as the one in #8 from the log?

Yes, everything is disabled on settings. Scanner mode balanced and monitor mode when device is connected.

d4rken, 2023-07-04:
I currently have no idea as to what could cause this.
Did you try just waiting a bit longer, maybe it just works, just takes longer until the data shows, and the troubleshooter is just a way to spend that time waiting.

silence-ong, 2023-09-04:
I'm on a POCO F5 running MIUI Global 14.0.6
I'm also encountering this issue. I have Monitor mode set to "when device is connected" and I've also set my Airpods Pro at the device address. After connecting my Airpods, CAPods notification pops up saying no device.
I've to press troubleshoot to find my device, like ceifa's case, the settings always revert back to normal. And if I try reconnecting my Airpods afterwords, everything works fine.

I've read in Openpods that they currently have issues working with MIUI, maybe that is a cause?

anderlopezsv, 2023-10-18:
Good afternoon. I have that problem too. I have a Samsung s21 FE, "No device" appears in the notifications and I can't find how to solve it.
[image]

anderlopezsv, 2023-10-18:
Good afternoon. I have that problem too. I have a Samsung s21 FE, "No device" appears in the notifications and I can't find how to solve it.

(I speak Spanish, sorry for any translation errors).

[image]

anderlopezsv, 2023-10-18:
Good afternoon. I have that problem too. I have a Samsung s21 FE, "No device" appears in the notifications and I can't find how to solve it.

(I speak Spanish, sorry for any translation errors).

[image]
[image]

d4rken, 2023-10-28:
Not much I can do, these are mostly ROM issues. Buggy Bluetooth Software Stacks.

Closing this in favor of #124

closed 2023-10-28 by d4rken