d4rken-org/capod #139: Add support for Borofone BW03

iHackFX, 2023-06-22:
Hello. I have these headphones, and here is what data I can provide. If there is a way I can help more, please tell me. Borofone BW03 it is Fake AirPods Pro.

| Event | Adv. packet |
|--------|--------|
| Case open | 0x18FF4C000713010E2075AA3732001000E4E4640000 |
| Case closed | 0x14FF4C00070F000E20B54FA7ACDCD995E4E4640900 |
| The right earpiece was pulled out of the case | 0x18FF4C000713010E2075AA3733001000E464640000 |
| The left earpiece was pulled out of the case | 0x18FF4C000713010E2075AA373300100064E4640000 |
| Music on both earpieces | 0x18FF4C000713010E2075 AA373300100064E4640000 (not sure)|

iHackFX, 2023-06-22:
And maybe i catch the Call State: 

```
0x14FF4C00070F000E20B54FA7ACDCD9956464FF0C00
```

d4rken, 2023-06-23:
On a first glance I think I spot the battery status `AA` in your example data.
And the start of the message `07 13 01`, this should be followed by the device identifier. In your examples that is `0E 20` which maps to AirPods Pro.

So far this all seems to match, where does this differ causing it not to be picked up by CAPod? You could copy the unit test for FakeAirPodsPro and copy your data into it and see what is decoded wrong:

https://github.com/d4rken-org/capod/blob/afa4d6e8071578da220435f3a83dca679f9e65ef/app-common/src/test/java/eu/darken/capod/pods/core/apple/misc/FakeAirPodsProTest.kt#L13-L38

iHackFX, 2023-06-23:
CAPod seems to find and display them when the setting is enabled "Unfiltered data BLE", but does not correctly read the state of the earpiece. In the example, I indicated that the state changes from E4 to 64, while the application displays that they are charging.

> You could copy the unit test for FakeAirPodsPro and copy your data into it and see what is decoded wrong

Honestly, I don't know how to work with Unit Test. I tried to do something for Android earlier but I never did anything.

iHackFX, 2023-06-23:
If it comes in handy, I am attaching the log file:
[image]

closed 2025-05-14 by d4rken