d4rken-org/capod #140: Add icon for Airpods Max

ApexioDaCoder, 2023-06-23:
It would be nice if instead of just displaying a generic headset icon, to show an icon that looks like the Airpods Max, just like the icon for the Airpods Pro.

d4rken, 2023-06-23:
Can you create one? I'm not good at doing icons.
Can't be one taken directly from Apple :skull:.

ApexioDaCoder, 2023-06-23:
Maybe you could use the one that OpenPods uses?
https://github.com/adolfintel/OpenPods/blob/master/app/src/main/res/drawable/podmax.png

bernzrdo, 2023-07-13:
@d4rken I can try creating one! What is the file type needed and size?

ApexioDaCoder, 2023-07-13:
iirc the file type is svg

d4rken, 2023-07-13:
An Android compatible SVG would be great (Android does not support every SVG feature), otherwise a PNG works too.

For most cases 192x192 is fine, if you make it 512x512 we have more room for the future. I'll downsize as needed.