d4rken-org/capod #154: Unable to use on Samsung 5 watch

yakov116, 2023-07-17:
When I open the app it requests notification permissions but the watch does not allow to grant it.

d4rken, 2023-07-17:
What prevents the user from granting it? Is the button greyed out? Not clickable? Is there an error?

Overall I'm dropping support for the WearOS app though :crying_cat_face:.
I've unpublished it from Google Play already and only the APK in GitHub releases remains.
I find that I don't have enough time or motivation to maintain the WearOS app.
Maybe someone else is interested in fixing this?

yakov116, 2023-07-17:
Its grayed out, this only started happening 3 versions ago.

While I totally understand you for wanting to drop support for wear os, I really enjoy your app and it works (or worked) really well.

d4rken, 2023-07-17:
> Its grayed out, this only started happening 3 versions ago.

Hm 3 updates ago I made the changes that Google required with the different splash graphic and time display: https://github.com/d4rken-org/capod/releases/tag/v2.9.3-rc0

If you side load that, does it work? Since I'm no longer uploading it to Google Play, I could revert all the changes I made just for Google's policies... :roll_eyes: 

> While I totally understand you for wanting to drop support for wear os, I really enjoy your app and it works (or worked) really well.

I'd love to keep it, it's mostly the Google Play related stuff being a headache. Google shouldn't be suprised why there are so few WearOS apps if they are so petty...

The code-base is already split into `app-common`, `app` and `app-wear`. Adding new devices happens just in `app-common` and does not require specific changes to the wear-os code, so currently maintenance is low to keep the WearOS APKs on GitHub.

Sideloading WearOS is not great though. Is it worth it? Would you sideload the WearOS APKs?  If we find someone to keep an eye on the WearOS builds and maintain it, then we could keep it around. Otherwise I'd keep it as long as it works, and remove it once it starts breaking.

What do you think?

yakov116, 2023-07-17:
I really understand your frustration, and your not the only one.

Its really a big pain to sideload, buy hey as long as it works I would just stay on that version.

I will try this later and get back to you.

Thanks

Macusercom, 2023-11-09:
I switched from the Fossil Gen 6 to the Samsung Galaxy Watch 5 and now I'm facing the same issue. It wants notification permissions but tapping the button does nothing. It looks like it's trying to open the settings panel but it doesn't exist on the smartwatch itself. In the Galaxy Wearables app the permission is enabled but that doesn't seem to change anything.

The error is:
`11-09 14:32:39.141  2677  2677 E GrantPermissionsViewModel: None of [android.permission.POST_NOTIFICATIONS] in {android.permission-group.LOCATION=[android.permission.ACCESS_BACKGROUND_LOCATION], android.permission-group.NEARBY_DEVICES=[android.permission.BLUETOOTH_CONNECT, android.permission.BLUETOOTH_SCAN]}`

closed 2024-06-26 by d4rken