d4rken-org/capod #164: AirPods Pro 2 usb c not supported

karimz1, 2023-09-23:
Do u know how to add that support to the app?

tsia, 2023-09-23:
[capod_logfile_1695485819291.log.zip](https://github.com/d4rken-org/capod/files/12706865/capod_logfile_1695485819291.log.zip)
In case this is helpful

d4rken, 2023-09-23:
Curious that Apple would change the device-id just for changing the case port?

d4rken, 2023-09-23:
@tsia Good log! Contained one data packed that was detected as unknown apple device by CAPod.

The identifier for AirPods Pro2 is `0x1420` and for the USB-C variant it seems to be `0x2420`.

The data has the right format and using the decoding for AirPods Pro2, the test-case I created gives out sane values:

https://github.com/d4rken-org/capod/blob/f6702dd595a7715bc256fcd618422823dbc271d3/app-common/src/test/java/eu/darken/capod/pods/core/apple/airpods/AirPodsPro2UsbcTest.kt#L29-L41

Does this correspond to the state the your pods were in when you recorded the log?

I'm going to release an update so you can test it.

The old AirPods Pro2 had trouble with in-ear detection, see #38.
As Apple changed the software for the USB-C variant :exploding_head:, it will be interesting to see if they fixed this now.

tsia, 2023-09-23:
Might be because the new ones do the vision pro stuff I think?

tsia, 2023-09-23:
Test case result looks fine to me though

d4rken, 2023-09-23:
Update should be available in ~10min:

https://github.com/d4rken-org/capod/actions/runs/6285657555

tsia, 2023-09-23:
Yup works now. In-ear detection is still weird though. In-ear is detected but taking it out is never detected. Need to put them in the case for the pod to now longer show as in ear.
Sometimes I see them appear twice in the app as well. One "right" and one "left" when I have debug mode enabled.

[capod_logfile_1695500111506.log.zip](https://github.com/d4rken-org/capod/files/12707232/capod_logfile_1695500111506.log.zip)
Here is another log while putting them in, taking them out and doing some other stuff. Maybe that helps

d4rken, 2023-09-24:
> Yup works now. 

:+1: 

> [capod_logfile_1695500111506.log.zip](https://github.com/d4rken-org/capod/files/12707232/capod_logfile_1695500111506.log.zip)
Here is another log while putting them in, taking them out and doing some other stuff. Maybe that helps

I extended the tests with this, thanks!

> In-ear detection is still weird though. In-ear is detected but taking it out is never detected. Need to put them in the case for the pod to now longer show as in ear.
Sometimes I see them appear twice in the app as well. One "right" and one "left" when I have debug mode enabled.

Sounds like #38, seems Apple did not fix this :frowning:.

PR #167 (2023-09-24): Extend test case for AirPods Pro 2 (USB-C)
Closes #164
commit 3f98c62a (2023-09-24): Extend test case for AirPods Pro 2 (USB-C)

Closes #164
closed 2023-09-24 by d4rken