d4rken-org/capod #176: The app can't detect the airpods

jasnn72, 2023-11-11:
Hello. I'm using a Vivo Y02s and have tried other apps (such as MaterialPods). Before, MaterialPods could see the percentage but now it can't (only that it is connected). The rest of the apps (including CAPod) can't detect my earphones when they are connected.
Tried some compatibility settings but it won't work. Stuck to guessing what the percentage is.

I'm using:
Airpods Pro 1st Gen

Adonis12aa, 2023-11-12:
Hola buenas tengo unos 2 generación

Adonis12aa, 2023-11-12:
Y no me conecta con niun dispositivo

jasnn72, 2023-11-12:
Hello, I tried it on a different phone and it connected the first time but the rest it didn't. I had to troubleshoot it first. After I closed the case it wouldn't connect anymore.