d4rken-org/capod #180: Auto pause doesn't work

TboythedoN, 2023-11-22:
The ear detection works and I have auto pause on but the auto pause doesn't work.

d4rken, 2023-11-28:
Pods? Version? Phone? Player? :question: :question: :question:

d4rken, 2023-11-28:
Probably https://github.com/d4rken-org/capod/issues/38

xiaoyaomeng, 2023-12-12:
很高兴收到您的邮件，看到后将立即回复~Thank you~

HK-DDboy, 2023-12-12:
这是一封自动回复邮件。已经收到您的来信，我会尽快回复。

ohadorpaz, 2023-12-23:
Same here.
Samsung galaxy 22, airpods pro Gen 2

filonitta, 2024-01-24:
If the feature doesn't work, and the issue seems out of scope, then I shouldn't buy the app. At least for now, keep watching the status

sohamsalhotra, 2024-02-25:
Same issue for me too.
Phone: Pixel 7 Pro / Android 14 Feb 2024 Update
App Version: v2.12.1-rc0
Device: AirPods Pro Gen 2 (USB-C) (Model No. A3048)
Music Source Apps: Multiple (Tested Spotify, YouTube, Google Chrome)
In-Ear status detected on App: Updates both ways within ~0.5s as seen on the App's main status screen (when inserted or removed; each side updates individually)

My current workaround: Audio is paused for each side when placed in the case, or for both sides when either is inserted in the ear and then paused via force sensor (Or, of course, manually pausing from the phone).

Doublemyst, 2025-01-05:
Same issue, any updates if this is something, which is going to be fixed?
In my case when i take the earbud out, it changes to ambient mode, but in the app it still says that is plugged in, but only when it is in music mode. when i have no music playing and take one earbud out, it recognize it, as one bud out of ear.