d4rken-org/capod #183: Can detect beats studio 3's but not AirPod Pro 2's

jonathanrobichaud4, 2024-01-12:
I recently bought a cheap Chinese android phone (which probably explains this issue) just to mess around with an see how I like android and I was trying to set the app up but it can't seem to detect my Airpod pros however, it works just fine with my beats. Do Beats Studio 3's communicate battery data over BLE like Airpod's? Or are they different?

d4rken, 2024-01-15:
> Do Beats Studio 3's communicate battery data over BLE like Airpod's? Or are they different?

Shouldn't be different, from the app-code side, the data is received the same, only what is in the data would be different. Strange behavior indeed :shrug:

jonathanrobichaud4, 2024-01-16:
Yeah it was super weird. I returned the phone just because I couldn't get it to work and I'm pretty sure they were spoofing support for the BLE scanning and/or it wasn't properly implemented. I'm on the lookout for a used pixel 6 now which will definitely be better all around! I'm going to close this as it wasn't an issue with CApod.

closed 2024-01-16 by jonathanrobichaud4