d4rken-org/capod #187: Auto pause does not work

Barty94, 2024-02-19:
Auto pause of media does not work, also ear detection not working when you are listening to media.

Device: Samsung Galaxy s23
Media player: Spotify

Issue: When listening to media, app does not detect when earbud is in ear, when you stop media and earbud is out of your ear, then it detects it as out in the app, when put back to ear, app detects it is back in ear and resume playing media again, so auto-play feature works properly, but auto-stop does not.

Please fix it as I purchased Pro version due this functionality.

Barty94, 2024-03-21:
Can you check this pls? I suppose it Will be easy fix, just some condition is not working as auto play and ear detection works fine.

d4rken, 2024-07-12:
Can you provide a debug log of this behavior?

On first glance, the play/pause code looks correct:

https://github.com/d4rken-org/capod/blob/main/app-common/src/main/java/eu/darken/capod/reaction/core/playpause/PlayPause.kt

d4rken, 2024-07-13:
@SimplyAnotherAnon That sounds a bit like #38 which is a bug in Apple's firmware 🙁

closed 2025-05-14 by d4rken