d4rken-org/capod #198: Notification shows wrong text when minimized

heschy2, 2024-03-15:
The Phone is connected to my Airpods Pro, so CAPod creates a persitent notification (see image):
[image]

However, the notification is minimized by default (which is okay and may be my fault). Thats not the problem, the problem is that the Minimized nitification says there is no device connected:
[image]

Seems to be a bug that (I assume) can be fixed easily, right (not meant to be a reproach I am just curious)?

d4rken, 2024-03-18:
What phone and ROM is that? Can't reproduce this on my device.

heschy2, 2024-03-22:
Fairphone 5 (FP5) with the newest Version of /e/OS

hf02, 2024-05-03:
Also getting this on a Galaxy S23 running One UI 6.1

TipzRickyCheung, 2024-06-26:
Reproduced on LineageOS 19.1 with OPPO A57

commit 07a11799 (2024-06-27): Implement support for different states in notification text

Fixes #198
closed 2024-06-27 by d4rken