d4rken-org/capod #202: My Device not found

Xvirted, 2024-05-06:
Hi I am trying to connect my Airpod Generation 3 airpods to my lhone SM-A037U, Android 13, and I have been trouble shooting and it keeps detecting Beats Solo 3 and I don't have any of those near me or anywhere around me and my Airpods aren't connecting at all. I will like to ask if your able to help me in any way please.

[capod_logfile_1714937192984.log.zip](https://github.com/d4rken-org/capod/files/15217516/capod_logfile_1714937192984.log.zip)

d4rken, 2024-07-12:
Is there a mixup in the detection? and are you sure it says "Beats Solo 3"?

The device code for AirPods Gen3 is currently noted as `0x1320`:

https://github.com/d4rken-org/capod/blob/051677e7dd38500bf57bc7024630fbbffb73507d/app-common/src/test/java/eu/darken/capod/pods/core/apple/airpods/AirPodsGen3Test.kt#L15-L17

your log file shows messages from with a device code of `0x0320` which are tracked as "Power Beats 3":

https://github.com/d4rken-org/capod/blob/051677e7dd38500bf57bc7024630fbbffb73507d/app-common/src/test/java/eu/darken/capod/pods/core/apple/beats/PowerBeats3Test.kt#L14-L16

d4rken, 2024-07-12:
Can anyone with AirPods Gen3 share a log file so we can check that `0x1320` is the correct identifier?

d4rken, 2024-07-12:
Your log also contains various types:

Don't know this one with `0x1120`

> 07 19 01 11 20 34 F9 3A 0D 00 00 18 F3 DD 9F 79 08 D7 53 69 08 13 B6 49 CA B2 31

or this one with `0x1720`

> 07 19 01 17 20 00 03 00 06 09 04 EF 1B B9 37 B2 07 A7 C0 0B 94 D5 10 B4 E8 3A DC