d4rken-org/capod #206: Feature Request: Don't connect to Airpods when not in ear

kavishdevar, 2024-06-04:
Hey!

It would be great if the app, using the data from airpods, determine if the airpods not in ear and/or are charging and prevent Android from connecting to it or change the audio output.

d4rken, 2025-05-14:
AFAIK, it's not possible to prevent the connection. Happy to re-open this if someone has more info.

kavishdevar, 2025-05-14:
there's a hidden disconnect() on BluetoothProfile. I use it in LibrePods, fwiw. It's kinda buggy, Fluoride is very confused when A2DP is not connected, and it keeps on trying to connect to it again. I disconnect it if the airpods are not worn as soon as Android connects.

d4rken, 2025-05-14:
Oh interesting! Thanks for the hint.

> there's a hidden disconnect() on BluetoothProfile.

And it's just callable, or only with root? ADB?

> I use it in LibrePods, fwiw. It's kinda buggy, Fluoride is very confused when A2DP is not connected, and it keeps on trying to connect to it again. I disconnect it if the airpods are not worn as soon as Android connects.

Do you the keep listening to connection events and then call `disconnect()` again?

Is that fast enough for the user to not notice a connection change (e.g. audio playback or UI affected)?

kavishdevar, 2025-05-14:
> And it's just callable, or only with root? ADB?
I use HiddenApiBypass; no special permissions needed.

> Do you the keep listening to connection events and then call disconnect() again?

yep, unfortunately

> Is that fast enough for the user to not notice a connection change (e.g. audio playback or UI affected)?

It's not that great. And if the A2DP profile is not connected for a long time, then for some reason, my app doesn't disconnect- I haven't looked into it yet, because I really am focusing on deassembling the macOS bluetooth stack to implement more features. (I might have stumbled upon Hearing Aid/Protection- a geoblocked feature :D)

> Cool project, your Librepods! Power of root 😁

Thanks! I wish there was a simpler way; like just updating the "Bluetooth app" because the Bluetooth App itself packs the library.

>  I wish Google would just fix the Bluetooth stack though 😦

+1!!

d4rken, 2025-05-14:
> I use HiddenApiBypass; no special permissions needed.

I wonder if that could get a dev into trouble on Google Play 🤔  

> I really am focusing on deassembling the macOS bluetooth stack to implement more features. (I might have stumbled upon Hearing Aid/Protection- a geoblocked feature :D)

That sounds awesome 🚀 👍

kavishdevar, 2025-05-14:
> I wonder if that could get a dev into trouble on Google Play 🤔

Probably... Maybe reflection could work? 

I wonder if i'll ever be able to get my app on there. Even OpenPods wasn't allowed. My app's core is literally to use hidden api to create the l2c socket. Maybe another report on issuetracker might get us public apis for using l2cap over br/edr.