d4rken-org/capod #208: Case Battery level not available airpods 2 pro Galaxy S20 FE

tgrandpierre, 2024-06-25:
Hello,
Congratulations for this great application! 

Everything is almost perfect but I can't get the case battery level  of my airpods 2 pro (lighting connector) bought 1 years ago on my Samsung Galaxy S20 FE (SM G780G model, Version One UI  5.1),  on android 13.

The battery level of the case  (icon) just say "not available". Both airpods level are ok.

I tryes to click several options without success,
Do you have an idea ?

Best regards

tgrandpierre, 2024-06-25:
I close the subject because for the first time since several months, it works great !b(I did nothing particular...)
So thank you for this great app !

tgrandpierre, 2024-06-25:
in fact it is not stable : case battery is no more available while airpods are...

d4rken, 2024-06-26:
Note that the case does not have it's own antenna. Case battery infos are only available if at least one AirPod is inserted into the case.

closed 2024-07-12 by d4rken