d4rken-org/capod #221: [Bug/Feature Request?] Fake AirPods "Pro 4 TWS"

moriel5, 2024-07-18:
Device: Razer Phone 2
OS: Android 9 - Stock

I had recently purchased (via AliExpress's Grow to Get program, otherwise I would not have bothered, as the battery is not user-replaceable) a set of fake AirPods, and attempting to use CAPods with them was odd, to say the least.

At first they were immediately detected, as fake Gen3 AirPods, except that only one earpiece was detected by CAPods (the case was not detected either, even if it should be detected), and CAPods thought that it was in call mode (airplane mode was on, by the way, with no calling apps running), however running diagnostics for not help, and instead CAPods only sometimes detected the earphones as compatible devices (or device, as the same issues as in the start happened, except that it was being detected as Gen2 instead of Gen3), while most of the time was unable to detect the earphones as compatible devices at all, even if I set the device address from the settings.

Additionally, it turns out that Razer's ROM supports the AAC Bluetooth codec, however when CAPods is running, the OS stops recognizing the AAC support, until I force close CAPods and reconnect to the earphones (sometimes two reconnects are necessary to restore AAC support).

Otherwise, the earphones (both) are working as expected, though I have yet to test all of the gestures (there seems to be inaccuracies in the manual, as tapping and holding for two seconds turns off or on the respective earpiece instead of doing anything on the list.

Here are the addresses, as provided by Gadgetbride (when logs are necessary, I shall provide them):
TWS, 41:42:4C:88:A4:9D, 00000000-0000-1000-8000-00805f9b34fb, 0000110b-0000-1000-8000-00805f9b34fb, 0000111e-0000-1000-8000-00805f9b34fb, 0000110e-0000-1000-8000-00805f9b34fb

d4rken, 2024-07-22:
> At first they were immediately detected, as fake Gen3 AirPods, except that only one earpiece was detected by CAPods (the case was not detected either, even if it should be detected), and CAPods [...]

The device address and type only help CAPod recognize the right headphones as yours. The information comes from BLE advertisements which are broadcast to everyone by AirPods in any case. So if the values change from one time to another that is just the pods sending different data for some reason.

> Additionally, it turns out that Razer's ROM supports the AAC Bluetooth codec, however when CAPods is running, the OS stops recognizing the AAC support, until I force close CAPods and reconnect to the earphones (sometimes two reconnects are necessary to restore AAC support).

See above, as CAPod never really interacts withe the pods (reaction settings aside), this seems to be a bug in the ROM from razer, if listening to BLE advertisements affects AAC support :shrug:.

> Otherwise, the earphones (both) are working as expected, though I have yet to test all of the gestures (there seems to be inaccuracies in the manual, as tapping and holding for two seconds turns off or on the respective earpiece instead of doing anything on the list.

> Here are the addresses, as provided by Gadgetbride (when logs are necessary, I shall provide them):

TWS, 41:42:4C:88:A4:9D, 00000000-0000-1000-8000-00805f9b34fb, 0000110b-0000-1000-8000-00805f9b34fb, 0000111e-0000-1000-8000-00805f9b34fb, 0000110e-0000-1000-8000-00805f9b34fb

Good for future features. :pray:

moriel5, 2024-07-22:
Thanks for the insights.

While the codec recognition certainly can be due to bugs in the ROM (there are probably bugs introduced by Razer in the WiFi and cellular modem drivers, after all, from additional testing, it seems to either be a bug with the earphones' firmwares, or an intentional feature misimplemented (again, in the earphones' firmwares), and is likely to be behind the issues getting CAPods recognizing the earphones.

By the way, since my post, I have not managed to get CAPods to recognize the earphones again at all.

Should I use nRF Connect to try gleaning more information?

NurMiApp, 2024-08-31:
> Edit: I just realized that I read the software name wrong, so henceforth I shall type "CAPod" instead of "CAPods".

OpenPods that's Windows available MagicPods for CAPod. Really this OpenPods? (Life at DotPod) :dependabot:

moriel5, 2025-04-14:
@NurMiApp I fail to understand?
My main OS on my desktop is Linux (Solus), not Windows, and we are talking about Android here, with CAPod.