d4rken-org/capod #223: How to build unsigned foss APK?

IzzySoft, 2024-08-03:
Is there a way I can tell gradlew just to build the foss flavor (preferably even without the wear component) without signing? Using `assembleFossRelease` it insists on having a keystore (despite having removed the corresponding parts from `build.gradle.kts` via `sed -r '/signingConfigs\[/d;/setupCredentials/d' -i app/build.gradle.kts` prior to calling `./gradlew assembleFossRelease`). And just calling `./gradlew buildFossRelease` doesn't give any APK.

Just trying to approve Reproducible Builds here. I'm no Android dev :wink: 

Thanks in advance!

d4rken, 2024-08-04:
All builds must be signed AFAIK, but the "debug" variants are signed by a default certificate set-up on installing Android Studio / tooling.

If it it does not need to be the release variant, `./gradlew :app:assembleFossDebug` should work (`:app:` prefix excludes the wear variant).
`./gradlew tasks` gives you an overview.

IzzySoft, 2024-08-04:
> All builds must be signed AFAIK

for distribution, yes. But for proving RB, we do not distribute any APK. And I don't have your keys, so I cannot sign it anyway (if I'd use different keys it would be a different APK again). For Reproducible Builds we build the *unsigned* APK, then use `apksigcopier` to copy over the signing blocks. If the build was identical, that APK should verify – if not, verification fails. At the end of the verification process, the APK is dropped anyway, so no need to sign even.

> but the "debug" variants are signed by a default certificate set-up on installing Android Studio / tooling.

That doesn't help, as the debug variant would always differ from the release variant, so it does not make sense to compare the two.

> If it it does not need to be the release variant

It needs to be **exactly the same variant** as the one to check, sorry.

> `./gradlew :app:assembleFossDebug`

Ah, nice: so just prefix the project – I didn't know that. Will try that as soon as I know how to get it build the unsigned APK instead of complaining about the non-existent keystore :wink:

obfusk, 2024-08-04:
> for distribution, yes. But for proving RB, we do not distribute any APK. And I don't have your keys, so I cannot sign it anyway (if I'd use different keys it would be a different APK again). For [...]

Actually, we *do* sign the APK. That's the only way to create a bitwise identical reproducible build. We're just not using `apksigner` or `gradle` to create a fresh signature using a private key and addit to the APK. Instead we copy upstream's existing one instead -- which of course requires an identical unsigned APK to be successful :)

Also, prepending `:app` doesn't change anything in this case: `:app:assembleFossRelease` and `assembleFossRelease` should be identical here (of course we can't use debug as you say and the use of the keystore not being optional make it hard to generate an unsigned APK without patching out a lot of code).

IzzySoft, 2024-08-04:
> Edit: never mind. Missed the `app-wear`.

Yeah, and rang my bell: I vaguely remembered my last errors came from that one. So now I not only managed to successfully build it, but it's even RB! Here's my recipe:

```yaml
        build:
          - sed -r '/signingConfigs\[/d;/setupCredentials/d' -i app/build.gradle.kts
          - chmod +x gradlew
          - ./gradlew :app:assembleFossRelease
          - mv app/build/outputs/apk/foss/release/*.apk /outputs/unsigned.apk
```

So with the next sync, Capod will carry the green shield for "RB successfully verified" :smiley: 

[image]

(from staging, to be published with the next sync around 6 pm UTC)

Thanks for your help, both Matthias and Fay!

closed 2024-08-04 by IzzySoft