d4rken-org/capod #226: Add support for AirPods 4 ANC

colour93, 2024-09-23:
I'm not sure if these are the only logs needed. I tried switching ANC on and off after the logs start recording.
[capod_logfile_1727060822224.log.zip](https://github.com/user-attachments/files/17092365/capod_logfile_1727060822224.log.zip)

d4rken, 2024-09-23:
Do you mean switching the ANC on/off via app?

Any "write" operations to Airpods require a solution for #215 which currently seems impossible :frowning:.

d4rken, 2024-09-23:
Nvm, i read this wrong. There are 2 different AirPods Gen4 Models.

colour93, 2024-09-23:
> Nvm, i read this wrong. There are 2 different AirPods Gen4 Models.

I may not be expressing myself accurately because I'm using a translator. I hope to provide logs to help you identify AirPods 4, but I'm not sure if my method of obtaining the logs is correct. Thank you very much for your project.

d4rken, 2024-09-23:
The log was correct, I just had some trouble identifiying your AirPods because there are so many other devices in range.

I think this is yours then:

```
2024-09-23T03:07:50.997Z  I/CAP:Pod:Factory: Pod created: UnknownAppleDevice(identifier=Id(id=5da27927-d5da-4813-b863-cfec6949537d), seenLastAt=2024-09-23T03:07:50.994553Z, seenFirstAt=2024-09-23T03:07:34.439609Z, seenCounter=3, scanResult=BleScanResult(-75, 4C:FD:8F:EC:77:FD, 1505818804198473, 76: 07 19 01 1B 20 0B 9A 8F 10 00 04 43 DF EC 1D D3 F1 C3 F4 A1 9B 29 26 B9 E7 3A A0, proximityMessage=ProximityPairing.Message(type=7, length=25, data=01 1B 20 0B 9A 8F 10 00 04 43 DF EC 1D D3 F1 C3 F4 A1 9B 29 26 B9 E7 3A A0), reliability=0.5524, rssiAverage=-73)
```

that would make the identifier `1B 20`

So based on your logs and the logs from #225.

* AirPods 4 (no-ANC) `0x1920`
* AirPods 4 (with ANC) `0x1B20`

I will use the same decoding mechanism as for AirPods 3 and include initial support in the next update. Let me know if anything looks wrong.

closed 2024-09-23 by d4rken

ASOwnerYT, 2024-09-24:
Thanks for adding this so quickly! All seems to work well

PR #228 (2024-09-23): Add support for AirPods (Gen 4 with ANC)
Closes #226
commit 21b6cfc0 (2024-09-23): Add support for AirPods (Gen 4 with ANC)

Closes #226