d4rken-org/capod #231: Auto Pause Does Not Work Galaxy 21 Ultra

dopeboyglory, 2024-09-24:
This feature does not work. I upgraded to premium very disappointed that its not working 😕.

baelrayneVessdren, 2025-02-05:
Same issue. 

Device: S25 Ultra
AirPods: Pro 2 (Lightning)
CaPod Version: Pro

AIrPods are recognized immediately. Feature is activated. When I take one of the AirPods out of my ear the status does not update in the app (still In Ear) and sound keeps playing.

baelrayneVessdren, 2025-02-05:
Had the same issue. Had to force stop the app a few times to get it to work. 

In my settings > Reactions > Deactivated One pod mode and activated everything else. Took both pods out and force stopped the app. Re-started it, put pods in my ears and waited till the app showed me "In ear" for both. Took one out after and had to wait ~2 seconds for the music to pause.