d4rken-org/capod #236: Add Airpods Max 2024 support

RsMan-Dev, 2024-10-16:
Hi, wanted to use your app for my airpods max 2024, but was not working, i can give you all data you need, just ask me.

d4rken, 2024-10-16:
Similar to here, I need a debug log while the device is in range.

Or you turn on debug log and grab the raw data on-screen. If they are like the old AirPods Max, I just need the 4 digit hex code that identifise them.

Example:
https://github.com/d4rken-org/capod/issues/225

RsMan-Dev, 2024-10-16:
[image]
does this helps you?

RsMan-Dev, 2024-10-16:
if not, please give me steps to enable logs and what i need to do to get what you want.

d4rken, 2024-11-08:
@b4shtirk1n has added support for the new AirPods via #238 :partying_face: 

@b4shtirk1n Have you testet your changes with real AirPods Max? The device code here looks different in the screenshot. :thinking: 

Can anyone record a debug log with AirPods Max old or new nearby?

PR #240 (2024-11-08): Fix AirPods Max and AirPods Max (USB-C) support
@b4shtirk1n I think we swapped the device codes in the previous PR.

Closes #236
commit 6b922e88 (2024-11-08): Fix AirPods Max and AirPods Max (USB-C) support

Closes #236