d4rken-org/capod #243: Auto stop / start bugging after some time

Soupolait, 2024-12-10:
Hi, I use Airpods Pro and CApod, installed on F-droid.
After some time listening to something, it will be put alternatively on pause / play / pause / play / pause / play and so on. I then have to force CApod to stop and start it again for it to work.
Doesn't anyone else has had the same issue ?

d4rken, 2025-05-14:
Possibly the in-ear status flags not being correctly reported by the AirPods.
Would need a debug log of this to see if some timing checks could fix this on the app side.

Soupolait, 2025-05-21:
Hi, I tried the app again and it now works fine. It would surprise me that the bug came from the AirPods, as they worked fine on iPhones. 

I was using a fork of LineageOS though, might have played a part in the issue (I'm now back on my phone's default OS). I really should've included logs, sorry. If I encounter the issue again, I'll recreate a topic with full infos / logs