d4rken-org/capod #248: Rep/fake airpods pro 2 USB-C battery incorrect

TheKingLol, 2025-01-08:
This particular instance is confusing, my fake airpods pro usb-c model gets detected as a fake non usb-c (since there isn't an option for a fake pro 2 usbc) but reactions and battery levels don't work. 

In this case, I don't really care about reactions but I feel like i can't live without a working widget for battery. 

To try and solve this issue, I decided to download android studio and independently create a new class for kotlin to add my own raw hex data (since it wasn't the same as the hex in [here](https://github.com/d4rken-org/capod/blob/main/app-common/src/main/java/eu/darken/capod/pods/core/apple/misc/FakeAirPodsPro2.kt)) tried to get BLE data but failed (as i have no clue how), I still tried to compile but I got stuck there aswell as I have little experience with android development.

Now I'm not too sure how you can help me exactly so I'll add a screenshot below.
[image]

I also believe that if you manage to create a function in the app where people can add hex data to the function, they can add support for their devices without having you to recompile the app but, as I mentioned, I probably can't speak on the subject since I don't know what it takes to build such app.

Many thanks,
King.

d4rken, 2025-01-09:
Your screenshot shows that the raw data from your device contains the identifier `14 20`.

This is the official identifier for AirPods Pro 2 (non USB-C). The "fake" part is only detectable because the fake ones send data that is shorter than normal:

https://github.com/d4rken-org/capod/blob/1ebb2319d4ef78a5533ddd7a9151c2f87b7cdf82/app-common/src/main/java/eu/darken/capod/pods/core/apple/misc/FakeAirPodsPro2.kt#L137-L140

Hm, I don't think there is a solution here. Your pods are just sending the wrong identifier.

> I also believe that if you manage to create a function in the app where people can add hex data to the function, they can add support for their devices without having you to recompile the app but, [...]

Technically possible, but just adding the raw hex data is not enough. You'd need to map all the values too. In the end it would be too complicated and too much work for a handful of people that would use it. And many more people would use that don't know how to use it, be confused and cause support work.

> To try and solve this issue, I decided to download android studio and independently create a new class for kotlin to add my own raw hex data (since it wasn't the same as the hex in [...]

Copying another similar class is a good start. I usually then also copy the test for that class, add in the raw data that I got, and then adjust the test to see if the mapping makes sense or needs to be adjusted.

> I still tried to compile but I got stuck there aswell as I have little experience with android development.

Hm should compile if you just download Android studio and open the project, what error did you get?

TheKingLol, 2025-01-09:
> Hm, I don't think there is a solution here. Your pods are just sending the wrong identifier.
>
Ah, kind of a bummer to not be able to see battery which is completely understandable 
> Hm should compile if you just download Android studio and open the project, what error did you get?
>
I'm not home right now but in my search history I have the error I got and I think it would be usefull to walk you through what I tried.
I copied your project using the built in git in android studio, after going through a few hurdles of getting gradle to sync, e.g 
`Could not resolve com.android.tools.build:gradle:8.1.2.` by just installing java 19, I look through your commits on adding reps as a guide to adding my own, after that I added my hex values but I had no clue how to obtain BLE data to add to `shouldBe` values. I decided to skip that step and compile but I got `> 'compileFossDebugJavaWithJavac' task (current target is 17) but got java 19 or something like that`. Thinking about it, downloading java 17 and compiling it would probably work so I probably should of held off opening an issue

d4rken, 2025-01-09:
> > Hm, I don't think there is a solution here. Your pods are just sending the wrong identifier.
> Ah, kind of a bummer to not be able to see battery which is completely understandable

I have to rephrase that. The identifier has no impact on the battery levels. Though if the battery levels are not changing, then this is also an issue with the fake pods.

Or does the raw data ever change for the battery levels? In your screenshot the `AA` corresponds to 100% for left and right pods.

TheKingLol, 2025-01-09:
> Or does the raw data ever change for the battery levels? In your screenshot the `AA` corresponds to 100% for left and right pods.
>
The raw data doesn't change at all, L and R pods are 100% with case stuck on 70%

I am particularly confused because for some reason a secondary iOS device I have shows correct readings though. I can be sure because when my case is 20% or lower, I see an orange light on the case instead of green or white

 I have no clue how iPhones communicate with real or fake airpods though which is why I thought maybe capod would be the app to check battery levels since it has real and fake device selections in settings.

Sorry for the inconvenience.

d4rken, 2025-01-17:
> I am particularly confused because for some reason a secondary iOS device I have shows correct readings though.

The data broadcasted via BLE consists of two parts:

* An unencrypted part, this is what we read on Android
* An encrypted part, this can only be read by Apple devices

AFAIK no one has figured out how to get decrypt the second part.
It's also possible that Apple just reads the data directy if the device is paired (this doesn't work on Android either, see #215)

closed 2025-01-17 by d4rken

TheKingLol, 2025-01-18:
Thanks for the insight, I wish you the best of luck on fixing other issues