d4rken-org/capod #287: [Feature request] use IRK to identify device

kavishdevar, 2025-05-19:
Because apple uses Resolvable Private Addresses to broadcast, we can use an IRK extracted from a Mac/Linux to resolve the RPA and identify the user's device. This can be helpful and when we want to show the case popup, and it can simplify the scanning process too.

d4rken, 2025-05-19:
How is that done? Got any documentation?

Can't be done directly on Android due to L2CAP protocol not being compatible?

kavishdevar, 2025-05-19:
> Can't be done directly on Android due to L2CAP protocol not being compatible?

Yeah, it is obtained over AACP.

> How is that done? Got any documentation?

Here, have a look at [this](https://github.com/google/bumble/blob/main/apps/ble_rpa_tool.py)!

kavishdevar, 2025-05-21:
Additionaly, using an encryption key, we can extract accurate battery information from the advertisment payload too! The data is encrypted using AES (ECB).

After decrpyting, here's what the data looks like:

```
14 e3 e2 32 0d2d73c2055b0003f2b2cb4b
```

All I know currently is that the bytes one, two and three are the battery levels and status. the bit \#0 shows if it is charging. byte 1 is primary pod, byte 2 is secondary pod, byte 3 is case.

In macOS's keychain, the IRK'll called `MagicAccIRK`, and the encryption key is called the `MagicAccEncKey`.

d4rken, 2025-05-28:
Going to test this soon. Sad that this is very much a very power user feature, getting that decryption key via MacOS is quite the hurdle 😔.

Can we use this to identify the airpods, does wrong decryption key throw an error or is it just random data,

d4rken, 2025-06-01:
Nvm, found it, the entry called `MobileBluetooth`, this is a good guide to get it:

https://www.home-assistant.io/integrations/private_ble_device/#getting-your-identity-resolving-key-irk

d4rken, 2025-06-05:
Works well 🍻

Did you have insighthow this works with beats or airpods max, i.e. "single pod devices", that only have one battery level?

kavishdevar, 2025-06-05:
> I had the same idea, on my TODO 😁

:D

> For unencrypted, e.g. AirPods Max, are using the data fields of the left pod. And the "are pods flipped" flag to indicate if the AirPods Max are being worn

Oh, I see! Thanks!

> I will add logging and make a pinned issue asking users to help.

That'll be great!

d4rken, 2025-06-06:
> Oh, I see! Thanks! I was thinking if it is just not used the same would be with the encrypted data.

Good point though, I have coded it like that now, we'll see what others report 🤷

PR #288 (2025-06-06): Support AirPods Identity-Resolving-Key and Encryption-Key 
Allow users to supply the identity-resolving-key and also the encryption-key. 

* The identity-resolving-key enables identifying AirPods despite address randomization.
* The encryption-key gives us access to some of the ecrypted data within the BLE broadcast.

Keys needs to be extracted (once) from the MacOS-keychain after pairing the pods to that computer.

Guide will be available here: https://github.com/d4rken-org/capod/wiki/airpod-keys

This implements a very cool idea from @kavishdevar, see #287.
Closes #287
closed 2025-06-06 by d4rken