d4rken-org/capod #290: Help wanted: Users with full headphones (AirPods Max, Beats Studio etc.)

d4rken, 2025-06-06:
Based on @kavishdevar in #287 support for decrypting and address resolving was added in #288 (available in v2.17.0-rc0+).

You can now provide CAPod with two things via settings:

* Identity Resolving Key (IRK). This key links the randomized Bluetooth addresses broadcasted by the AirPods. It allows CAPod to uniquely identify your AirPods, eliminating the need for heuristics or signal strength checks.
* broadcast Encryption Key. This key decrypts the private part of the broadcasted message, allowing CAPod to retrieve more accurate battery levels (e.g., 1% steps instead of 10%).

I've tested this for `AirPods Pro` and `AirPods Pro 2`. I'm confident that the logic is the same for all "dual pod devices", i.e. two pods, two batteries and a case battery. For "single pod device" I'm not so sure, i.e. AirPods Max only have one battery and no case, so the data may look different.

If you have such a device and want to help. Enter the keys into CAPod ([Guide here](https://github.com/d4rken-org/capod/wiki/airpod-keys) and see if they work. If it worked, then a key icon should be displayed, and the battery level should be displayed in 1% steps.

[image]

If the key icon is not displayed, or the battery values are wrong or don't make sense. Please post here and make a debug log available (either through Discord, by uploading here, or mailing to support@darken.eu).

d4rken, 2025-06-13:
> but I have a question too
this version will show other people's airpods
(with or without my airpods connect)
and it will say it's mine (show 'yours')
also it will show the key too(I definitely not set others IRKs)
maybe it still have bugs?

Possibly, can you open a new ticket with screenshots and maybe a debug log? I will look into it.
Showing others could be the case but they should not show as yours or show the key.
Are you maybe seeing your left and right pod individually?

closed 2025-06-13 by d4rken

d4rken, 2025-06-13:
Should fix your other issue too, if not, make a new ticket with a debug log.

sujiba, 2025-06-25:
In the first post there is a guide linked. In the guide you'll find a link to [Home Assistant Private BLE Device](https://www.home-assistant.io/integrations/private_ble_device/). Maybe this works for you.

kavishdevar, 2025-07-07:
@AnthonyLloydDotNet because you're on linux, you can directly communicate with AirPods and retrieve the keys from there itself.

here's a small script you can use to do so :)

```
import sys
import socket

PROXIMITY_KEY_TYPES = {
    0x01: "IRK",
    0x04: "ENC_KEY",
}

def parse_proximity_keys_response(data):
    if len(data) < 7 or data[4] != 0x31:
        return None
    key_count = data[6]
    keys = []
    offset = 7
    for _ in range(key_count):
        if offset + 3 >= len(data):
            break
        key_type = data[offset]
        key_length = data[offset + 2]
        offset += 4
        if offset + key_length > len(data):
            break
        key_bytes = data[offset:offset + key_length]
        keys.append((PROXIMITY_KEY_TYPES.get(key_type, f"TYPE_{key_type:02X}"), key_bytes))
        offset += key_length
    return keys

def hexdump(data):
    return " ".join(f"{b:02X}" for b in data)

def main():
    if len(sys.argv) != 2:
        print(f"Usage: {sys.argv[0]} <MAC>")
        sys.exit(1)

    bdaddr = sys.argv[1]
    PSM = 0x1001

    handshake = bytes.fromhex("00 00 04 00 01 00 02 00 00 00 00 00 00 00 00 00")
    key_req = bytes.fromhex("04 00 04 00 30 00 05 00")

    sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_SEQPACKET, socket.BTPROTO_L2CAP)
    sock.connect((bdaddr, PSM))
    sock.send(handshake)
    sock.send(key_req)

    try:
        while True:
            pkt = sock.recv(1024)
            keys = parse_proximity_keys_response(pkt)
            if keys is not None:
                print("Proximity Keys:")
                for name, key_bytes in keys:
                    print(f"  {name}: {hexdump(key_bytes)}")
                break
    finally:
        sock.close()

if __name__ == "__main__":
    main()
```

AnthonyLloydDotNet, 2025-07-10:
> [@AnthonyLloydDotNet](https://github.com/AnthonyLloydDotNet) because you're on linux, you can directly communicate with AirPods and retrieve the keys from there itself.
>
> here's a small script you [...]

that worked perfectly. thanks for the script legend

kavishdevar, 2025-07-11:
> I am unable to get the script working, am I doing something wrong? I am a noob when it comes to linux and my only linux device is a Steam Deck.

Try connecting your AirPods first and then running this script.

Delhpikus, 2025-07-11:
> > I am unable to get the script working, am I doing something wrong? I am a noob when it comes to linux and my only linux device is a Steam Deck.
> 
> Try connecting your AirPods first and then running this script.

Thanks for the reply, however they are already connected. When I run the script, it just hangs and nothing happens.

[image]

Delhpikus, 2025-07-12:
> > Thanks for the reply, however they are already connected. When I run the script, it just hangs and nothing happens.
> 
> That's werid. Do you have MagicPods running because that might be using the socket and preventing the script from running. Also, I think it'd better to create a new issue for this because it's going off-topic for this.

Yep, that was it. You're a legend. :)