d4rken-org/capod #326: Add support for Apple Airpods Pro 3

Zeratoxx, 2025-09-20:
Hi,

I just got the Apple Airpods Pro 3, do you need some information?

Leclowndu93150, 2025-09-20:
same, i'm a developer myself i know how to use cmd/tools i can help give reverse engineering information

d4rken, 2025-09-21:
#325,#327,#328 will add support.

Please test the next update and reopen this issue if it does not work.

closed 2025-09-21 by d4rken

jackfranklin, 2025-09-22:
> #328 <https://github.com/d4rken-org/capod/pull/328> is part of v2.19.0-rc1
> (or later), see: https://github.com/d4rken-org/capod/releases
>
> Your screenshots show v2.18.0-rc1.
>
> Google Play releases always take a bit longer. I just checked and you
> should see that version on Google Play in the next few hours.

Oh I apologize I misread the version number! Sorry for the noise.

PR #328 (2025-09-21): AirPods Pro 3 support
Merge between

* #327 by @atharen
* #325 by  @li-fstz

Closes #326